NModbus/NModbus #1: Documentation for adding custom function services to slave

sujeethk, 2017-05-02:
Is there a documentation for adding custom function services to the slave?

rquackenbush, 2017-05-03:
Hi - not yet. I just coded this (last week?) so I haven't had a change to create documentation for it. Is there a particular function service you're interested in (e.g File, etc)?

I'll start adding to the wiki soon, but the basic idea is that you'll have to pass an additional service implementation into the constructor for `ModbusFactory`.

You can look at the implementation of the built in handlers in the meantime: https://github.com/NModbus/NModbus/tree/NetworkedSlave/NModbus/Device/MessageHandlers

sujeethk, 2017-05-03:
@rquackenbush Good to see someone actively working on NModbus. I'm trying to have a custom service that will handle request based on access levels(basically hide a few registers from Read). After posting this question, I was able to sketch a Custom function and pass it to ModbusFactory with BuiltIn set to True. It works!

Still working through the code. I will get back with some feedback in the next few days. If you have a Gitter channel let me know.

Currently trying to figure out why my datastore isnt shared among different Clients/Masters.

sujeethk, 2017-05-03:
@rquackenbush on a sidenote, check your local Git config. Looks like it is not set to your Git account - the commits seem to be from "rich.quackenbush" instead of your Git account

rquackenbush, 2017-05-04:
@sujeethk  - thanks will do.

rquackenbush, 2017-12-07:
@sujeethk - I finally set up a gitter room:
https://gitter.im/NModbus

closed 2020-04-15 by rquackenbush