NModbus/NModbus #2: Are you planning to update the NModbus.Serial code for .NET Core?

dirkga, 2017-07-31:


rquackenbush, 2017-08-01:
I'm not sure... I'm not fully up to speed on .NET Core. But as I understand it, SerialPort support under .NET Core has been a contentious issue:

https://github.com/dotnet/corefx/issues/984

If a serial port implementation makes it into the framework, I will likely add it. Pull requests are always welcome :-)

ManniAT, 2017-09-28:
Hi,

I had the same requirement. For simplicity I copied your SerialAdapter - sourcecode into core 2.0 project.
Then I added https://www.nuget.org/packages/System.IO.Ports/ and it works like a charm.
I tried it on a rapspberry PI running windows-10 IOT, from a console APP as well as from a MVC Web app.
In both cases I used RTUMaster to retrieve data from HTC (humidity, temperature, carbon dioxide) device. It's not a problem to host the class myself. But it would be nice if you could update your nuget with a .NET core option.

Thank you for your great work with NModus!!

rquackenbush, 2017-09-29:
Nice! I'm glad that worked. I'll add that to NModbus soon.

ManniAT, 2017-10-01:
By the Way - I did a lot with it in the last days - and tests on full Windows10 PC (surface pro) also worked without any problems.

rquackenbush, 2017-11-13:
@ManniAT  any chance you could submit a pull request with your approach? I'd like to integrate it.

LGinC, 2017-11-13:
@ManniAT  
UWP or .net core Console Application?
SerialPort or SerialDevice？

rquackenbush, 2017-11-13:
@ManniAT @LGinC  I just added .NET Core support for the `NModbus.Serial` package as of version `3.0.37` (untested) Please give it a go and let me know if it works for you.

rquackenbush, 2017-11-13:
Marking this as closed lest I hear something back about any additional issues.

closed 2017-11-13 by rquackenbush

LGinC, 2017-11-14:
@rquackenbush Install NModbus.Serial 3.0.37 failed  in UWP

LGinC, 2017-11-14:
I write a SerialDevice Adapter

using System;
using System.Diagnostics;
using System.Linq;
using System.Runtime.InteropServices.WindowsRuntime;
using System.Threading;
using System.Threading.Tasks;
using Windows.Devices.SerialCommunication;
using Windows.Storage.Streams;

namespace IntelligentLocker.Helper
{
    /// <summary>
    /// UWP SerialDevice Modbus Adapter
    /// </summary>
    public class SerialDeviceAdapter : NModbus.IO.IStreamResource
    {
        private SerialDevice _serialDevice;
        private IInputStream inputStream;
        private IOutputStream outputStream;
        public int InfiniteTimeout => Timeout.Infinite;

        public int ReadTimeout { get => _serialDevice.ReadTimeout.Milliseconds; set => _serialDevice.ReadTimeout = TimeSpan.FromMilliseconds(value); }
        public int WriteTimeout { get => _serialDevice.WriteTimeout.Milliseconds; set => _serialDevice.WriteTimeout = TimeSpan.FromMilliseconds(value); }

        public SerialDeviceAdapter(SerialDevice serialDevice)
        {
            Debug.Assert(serialDevice != null, "Argument serialDevice cannot be null");
            _serialDevice = serialDevice;
            inputStream = _serialDevice.InputStream;
            outputStream = _serialDevice.OutputStream;
        }

        public void DiscardInBuffer()
        {
            _serialDevice.OutputStream.FlushAsync();
        }

        public void Dispose()
        {
            _serialDevice.Dispose();
        }

        public int Read(byte[] buffer, int offset, int count)
        {
            IBuffer tBuffer = buffer.AsBuffer();
            Task t = Task.Run(async() => await inputStream.ReadAsync(tBuffer, (uint)count, InputStreamOptions.None));
            t.Wait();
            tBuffer.ToArray().CopyTo(buffer, 0);
            return (int)tBuffer.Length;
           
        }

        public void Write(byte[] buffer, int offset, int count)
        {
            IBuffer tBuffer = buffer.AsBuffer();
            outputStream.WriteAsync(tBuffer);
        }
    }
}

rquackenbush, 2017-11-16:
@LGinC - thanks for that code.  I'm still on the learning curve for cross platform support in NuGet packages. Version `3.0.38` of the `NModbus.Serial` package has support for UWP (I even tested it!)

LGinC, 2017-11-17:
@rquackenbush oh....there is not any DiscardInBuffer API in SerialDevice class, so.....maybe this function can not used.

rquackenbush, 2017-12-07:
@LGinC - updated and available as of version 3.0.47 of the NuGet package.

I've also added `SerialPortStream` support (serial on .NET core running on Linux targets): 
https://www.nuget.org/packages/NModbus.SerialPortStream/