NModbus/NModbus #5: RTU over IP

oliverschenk, 2017-08-22:
Is there a way to create an IP connection but send Modbus RTU type packets? E.g. Modbus RTU Over IP?

We use quite a lot of old legacy Modbus serial devices but they are connected via IP wireless Modems that forward packets received on port 502 to the serial device. However it isn't Modbus IP it's ModbusRTU over IP.

I noticed that the TcpClientAdapter class is marked as internal, but if this was made public I would be able to do something like:

```
TcpClientAdapter adapter = new TcpClientAdapter(client);

IModbusFactory factory = new ModbusFactory();
_modbusMaster = factory.CreateRtuMaster(adapter);
```

rquackenbush, 2017-08-22:
Version 3.0.36 of the Nuget package now his this class (as well as `UdpClientAdapter`) marked as public. Please let me know if this works for your scenario.

closed 2017-08-22 by rquackenbush

oliverschenk, 2017-08-25:
The assembly doesn't seem to be strongly named. Would you consider doing this for the released NuGet package?

```
System.IO.FileLoadException: 'Could not load file or assembly 'NModbus, Version=3.0.36.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. A strongly-named assembly is required. (Exception from HRESULT: 0x80131044)'
```

CDDFERREIRA, 2019-06-03:
I am looking to perform similar operation.
I have a device that acts as a gateway converter from Ethernet to RS485 Serial. And, my field device is a Modbus RTU protocol type. 

I have tested connect the device directly and got it working with NMODBUS using the SerialMaster. Now I am looking at interfacing it through my bridge hardware that encapsulates an I.P. address and port number to a serial comm internally.
So its still a RTU Modbus device, but I am using Ethernet to connect to it. Any example help? I am bit lost how I can modify the NMODBUS classses to make it work.

Much appreciated.

gsulc, 2019-06-03:
@CDDFERREIRA, the feature was added so you can use it like the poster above:
```csharp
TcpClient tcpClient; 
var modMaster = factory.CreateRtuMaster(new TcpClientAdapter(tcpClient));
```
I currently use it, so it works.