NModbus/NModbus #6: Is there a way to expand the Modbus user specific function code ?

yazelin, 2017-10-23:
How do I build a Modbus Master and add a user specific function code (101 0x65) and SubFunction Code (1-10).
Is there a way to expand the Modbus user specific function code ?

rquackenbush, 2017-10-23:
I haven't done this myself yet, but it looks like you need to implement the `IModbusFunctionService` interface and pass this into the `ModbusFactory` constructor.

```c#
// Create an instance of your service (or services)
IModbusFunctionService[] services = new IModbusFunctionService[]
{ 
    new MyFunctionService()
};

// Create a ModbusFactory instance using your custom service(s)
ModbusFactory factory = new ModbusFactory(services, true);
```

yazelin, 2017-10-23:
so add this code can create Modbus slave and handle function code 101 From "Modbus master" ?

If i use modbus master send function code 101 to Modbus slave ?
did i need some class like that "NModbus\NModbus.IntegrationTests\CustomMessages\" ?

yazelin, 2017-10-24:
I SEE...

closed 2017-10-24 by yazelin

BullT, 2019-02-05:
Hi,

Could you explain how you did?
Or share an example?

Thanks,