NModbus/NModbus #8: Can I Use NModbus in Unity?

DarkDay8, 2017-12-19:
Hello!
I have two virtual devices written on Unity.  There are these programs on two computers connected to a com ports. It is necessary to implement them communication via Modbus ASCII protocol. Can I use NModbus? Will I get problems with multithreading?

rquackenbush, 2017-12-19:
Hi @DarkDay8 - I've never done any work with Unity. After a short search, it looks like Unity uses mono for c# support. If that's the case, then yes, NModbus should run just fine. 

In terms of multi threading, the library doesn't do any locking for the master. If you will have concurrent access to the ModbusMaster, you will have to add your own locking mechanism. One way to do this is to implement a shim of `IModbusMaster` that wraps one of the existing master implementations.

```c#
public class LockedModbusMaster : IModbusMaster
{
    private readonly IModbusMaster _inner;
    private readonly object _lock = new object();

    public LockedModbusMaster(IModbusMaster inner)
    {
       _inner = inner;
    }

    public void WriteRegisters(byte unitId, ushort address, ushort[] points)
    {
        lock(_lock)
        {
            _inner.WriteRegisters(unitId, address, points);
        }
    }
}
```

Also, be sure to implement a pause in between master requests so that the serial Modbus line is quiet for long enough.

DarkDay8, 2017-12-21:
Hi @rquackenbush - I'm interested in a slave device and her listen function. Whether there is a possibility correctly Create/Dispose slave device? Encountered a problem with disposing listen Slave device.

rquackenbush, 2017-12-21:
@DarkDay8 - what problem did you have? Are you able to attach some sample code to reproduce the problem?

DarkDay8, 2017-12-21:
Unfortunately, I could not connect your libraries to my project. Firstly, there is no support for C# 7 in the Unity. Secondly, I noticed 
using Windows.Devices.SerialCommunication;
using Windows.Storage.Streams;
They do not exist in mono.
Previously used NModbus_net-3.5.  You use the same name.

rquackenbush, 2017-12-21:
Are you compiling the code directly or referencing the nuget packages? If you reference the nuget packages, you can just pick the bits you want. The solution has the code to build for all supported targets, not just mono. 

For mono, I would suggest installing these two packages:

- https://www.nuget.org/packages/NModbus/
- https://www.nuget.org/packages/NModbus.Serial/

I have used those successfully under mono.

DarkDay8, 2017-12-22:
I can not clean up resources properly. When I re-create an object, I get a takeoff

public class SlaveANSII : MonoBehaviour 
{
    SerialPort _slavePort = new SerialPort("COM2");
    IModbusSlaveNetwork _network;

    private void Awake()
    {
        GameObject.Find("Slave_btn").GetComponent<Button>().onClick.AddListener(CreateSlaveDevice);
        GameObject.Find("Dispose_btn").GetComponent<Button>().onClick.AddListener(DestroySlave);
    }
    void Start ()
    {		
    }
    void Update ()
    {		
    }
    void CreateSlaveDevice()
    {
        if (_slavePort != null && _slavePort.IsOpen)
        {
            _slavePort.Close();
        }
        _slavePort.BaudRate = 9600;
        _slavePort.DataBits = 7;
        _slavePort.Parity = Parity.None;
        _slavePort.StopBits = StopBits.Two;
        _slavePort.Open();
        var factory = new ModbusFactory();
        var adapter = new NModbus.Serial.SerialPortAdapter(slavePort);
        var transport = factory.CreateAsciiTransport(adapter);
        _network = factory.CreateSlaveNetwork(transport);
        var slave = factory.CreateSlave(1);
        _network.AddSlave(slave);
        this.StartCoroutineAsync(Listen());
    }
    private IEnumerator Listen()
    {
        _network.ListenAsync();
        yield return null;
    }
    private void DestroySlave()
    {
        _network?.ListenAsync().Dispose();
        _network?.RemoveSlave(1);
        _network?.Dispose();  
        _slavePort?.Close();
    }
    private void OnDestroy()
    {
        DestroySlave();
        _slavePort?.Dispose();
    }
}
I need to can creating and destroying devices

rquackenbush, 2017-12-23:
What does "a takeoff" mean?

I did some digging into the Unity documentation:
https://docs.unity3d.com/ScriptReference/MonoBehaviour.StartCoroutine.html

It looks like it repeatedly calls the `Listen()` method. That could create some issues.

Below is another way to attach the problem. I was able to create and recreate the slave network at will.

        private void CreateSlaveNetwork()
        {
            if (_slavePort.IsOpen)
            {
                _slavePort.Close();
            }

            _slavePort.BaudRate = 9600;
            _slavePort.DataBits = 7;
            _slavePort.Parity = Parity.None;
            _slavePort.StopBits = StopBits.Two;
            _slavePort.Open();

            var factory = new ModbusFactory();

            var adapter = new SerialPortAdapter(_slavePort);

            var transport = factory.CreateAsciiTransport(adapter);
            _network = factory.CreateSlaveNetwork(transport);
            var slave = factory.CreateSlave(1);
            _network.AddSlave(slave);
            
            //This forces the listen method to execute on a background thread
            Task.Factory.StartNew(() => _network.ListenAsync(), TaskCreationOptions.LongRunning);
        }

        private void DestorySlaveNetwork()
        {
            _network?.RemoveSlave(1);
            _network?.Dispose();
            _slavePort?.Dispose();
        }