NModbus/NModbus #9: Nmodbus for Raspberry pi

Thonelson, 2017-12-19:
Hi

Im currently trying to implement modbus on raspberry pi with the Windows 10 iot core OS. Does Nmodbus support .NET core? If so, how does one publish a netcoreapp from this library to run on the pi?

rquackenbush, 2017-12-19:
@Thonelson - the library should run on Windows 10 IoT. There is some mention of this in this issue:

https://github.com/NModbus/NModbus/issues/2

I don't know anything about publishing to Windows 10 IoT.

CrosserTechnologies, 2017-12-26:
@Thonelson @rquackenbush Since `NModbus` is targeting `netstandard2.0` you should be able to just add the package and then build for your specific target by using the correct `RID` ([see catalog](https://docs.microsoft.com/en-us/dotnet/core/rid-catalog)).

We use RPi with dotnet core, but we run Linux, not Win10 IoT. However, it should work for you if you use the correct `RID`.

Something like this should point you in the right direction.

### .csproj
```
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <LangVersion>latest</LangVersion>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.0</TargetFramework>
    <RuntimeIdentifiers>win10-arm</RuntimeIdentifiers>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="NModbus" Version="3.0.48" />
  </ItemGroup>
</Project>
```

Then build for Win10-arm with this command

`dotnet build -r win10-arm`

Worked fine for me, however I did not try to move the result onto a actual Win10 IoT machine.