NModbus/NModbus #11: Writing slaves is complicated as one should custom implement ISlaveDataStore in order to respond to data changes properly

glmnet, 2017-12-28:
I realized after a while I had to use a custom implementation of ISlaveDataStore to react to changes in the DataStore, this is confusing specially because in previous versions of the library there were event handlers for this.
I believe a custom implementation of ISlaveDataStore that will fire and event *only when the change is initiated from a Modbus master write* (or otherwise include change event source in event arguments) should be provided.

rquackenbush, 2017-12-30:
I ended up going with a more bare implementation of the slave data store than was previously provided as I felt it was too limiting. 

I can see where providing an implementation of ‘ISlaveDataStore’ that more closely mimics the legacy data store would make the transition easier and reduce boilerplate code. I’ll add that soon.

joncutting, 2019-07-16:
I see the same issue as glmnet in a project that am I working on...has there been any progress here?

AlexanderNitsch, 2020-03-03:
I would love that as well. Did anyone already write a DataStore which has this feature? Some sample code to start would be nice. (An official release even nicer :) )

rquackenbush, 2020-04-15:
The `SlaveDataStore` class now exposes the underlying `PointSource<T>` instances (rather than `IPointSource<T>`). This includes the following events:

```c#
public event EventHandler<PointEventArgs> BeforeRead;
public event EventHandler<PointEventArgs<T>> BeforeWrite;
public event EventHandler<PointEventArgs> AfterWrite;
```

This will be included in the next release.

closed 2020-04-15 by rquackenbush