NModbus/NModbus #13: WriteMultipleRegister - max registers

FabioGenesy, 2018-01-16:
What is the max length of the data array on the WriteMultipleRegister command?
I use version 1.11 of the library.
From my experiments I can't write more than 48 registers.
Is this limitation removed in the newer version?

rquackenbush, 2018-01-18:
The max size is 127 registers:

https://github.com/NModbus/NModbus/blob/develop/NModbus/Modbus.cs#L9
 
I'm guessing that you're using one of the older NModbus4 packages. The latest version is 3.0.48:

https://www.nuget.org/packages/NModbus/

malkot, 2021-02-16:
Please tell me . There is the following scheme - 1 master and 2 slaves. The master interrogates the register of slave number 1 and, if necessary, writes the value to the register of slave number 2 by condition. Master and slave number 1 is hardware. Slave number 2 I'm trying to make software. The problem is as follows, if the master reads the register value from slave number 1, then when starting slave number 2, I get the following error 

** "Unhandled exception of type" System.ArgumentOutOfRangeException "in Modbus.dll

More information: Maximum amount of data 127 registers " **

If slave number 1 is not polled, everything works well.

```
namespace WindowsFormsApplication10
{
    public partial class Form1 : Form
    {
        private SerialPort slavePort = null;
        private ModbusSlave slave = null;
        public Form1()
        {
            slavePort = new SerialPort("COM20");
            slavePort.BaudRate = 9600;
            slavePort.DataBits = 8;
            slavePort.Parity = Parity.Even;
            slavePort.StopBits = StopBits.One;            
            slavePort.Open();
            byte unitID = 4;
            slave = ModbusSerialSlave.CreateRtu(unitID, slavePort);      
            slave.DataStore = DataStoreFactory.CreateDefaultDataStore();            
            Thread th1 = new Thread(Listen);
            th1.IsBackground = true;
            th1.Start();
            InitializeComponent();
        }
        public void Listen()
        {
            slave.Listen();
        }
        private void button1_Click(object sender, EventArgs e)
        {
            textBox1.Text = slave.DataStore.HoldingRegisters[1].ToString();
        }        
    }
}
```