NModbus/NModbus #14: TCP connection to master not closed  on dispose

mcroissant, 2018-01-24:
Hello,

I'm currently testing the library for modbus communication in a .Net Core app, however it appears that after disposal of TCPClient the connection to the master is not closed, at least that's what my simulator is telling me (it was closing with an other library : libmodbus C++).

Any advise on how I could get the connection to close properly ? any hint on where to look for, I would'nt be against providing PR.

rquackenbush, 2018-03-06:
Apologies for the (very long) delay getting back to you. Unfortunately, I don't do any Modbus over TCP in my work, so I don't have any experience in this area. I would definitely appreciate a PR!

Fuchks, 2018-06-06:
As NModbus simply uses an established connection over TcpClient and does not handle opening or closing the connection, I suggest that the problem is outside the scope of NModbus.

closed 2018-09-11 by rquackenbush