NModbus/NModbus #16: TCP Modbus problem

jbrzo, 2018-02-16:
There is a problem with TCP Modbus. In case when the responses come later, the communication comes to situation, when the received responses are older than the command message. Difference between request ID and response ID leads to exception. This exception causes the command message is repeated immediately again, and the communication is flooded.

The NModbus remembers only last command, so the responses with ID lower than command ID should be ignored, none ecxeption should be thrown, and the reading should be repeated. 
The exception should be thrown only when the response ID is bigger than command ID. 

J.B.

rquackenbush, 2018-03-06:
Apologies for the delay - any chance you could submit a PR for this? I don't have any TCP equipment here to test on.

jremillard, 2018-06-15:
I have lots of TCP modbus hardware, how can I reproduce this problem?

Fuchks, 2018-06-15:
@jremillard
The standard timeout for responses is set to 250ms.
If the response takes longer, NModbus sends another request and the issue takes place.
With real hardware it is hard to reproduce, as there has to be introduced a delay of the response by at least 250ms.
But with the tool [Ananas](https://www.google.com/url?sa=t&source=web&rct=j&url=http://www.tuomio.fi/ananas/&ved=2ahUKEwj3n9v2ptXbAhVB7xQKHTHjB00QFjAAegQIBxAD&usg=AOvVaw35ewlU9IVmbrGZmNBle63B) it is quite easy to reproduce: enable option reply out of sync.

closed 2018-06-27 by rquackenbush