NModbus/NModbus #17: Trio / Schneider 32bit words

cragdor, 2018-03-06:
Hi All, 

I've used NModbus4 in the past and i'm looking to use, NModbus from now on since NModbus4 appears dead(No commits for the last two years).

We use the library to talk to some server/stepper motors with build in TCP Modbus controllers. However the controllers work in 32bit words, mostly big endian, one little endian. They do not support 16bit words so 2 registers have to be read/written for each parameter.
 
In NModbus4 I created a new ModbusMaster called ModbusMaster32bit(transport, enum.Endian), which handled the conversions. Obvious that resulted in a lot of duplicate code, as the master was effectively performing the conversions and then just calling the stock implementation. 

I'm curious if you think there is a better way to do this(Perhaps with a wrapper class?). The conversion is simple for Trio and Schneider as they use Big endian Floats/UInts. Also there was a simple ASCII to HoldingRegisters converter because the serial/model numbers are holding register ranges(where one ushort = 2xchar).

I'd like to know if it is worth implementing the additions to the code base and the best place to do it? 
Or to keep it complete separate from NModbus.

Craig

Below is the extract from one of the manuals.

[image]

rquackenbush, 2018-03-06:
I have the same need in my applications. So far, I've been handling all of that externally in a wrapper class. I have the concept of a `IRegistryRange` that allows different implementations to be swapped in (real time read/write vs batch). I think that I should at least include some of this code in an example. If the community has a use for it, I can add this (perhaps in a separate nuget).

cragdor, 2018-03-08:
I have notices that @Maxwe11, has already made an implementation of this for LittleEndian, He has a class that wraps an IModbusMaster and performs the conversion to Short/Int/Long/UShort/UInt/Ulong/Float/Double/Float.

cragdor, 2018-03-08:
It would be good if the Modbus Enhancements could be added to the main project as i expect that a lot of people would use them.

cragdor, 2018-03-13:
FYI - I'm implementing a wrapper library called Modbus.Enhanced. It wraps and extends IModbusMaster, 
I have : 
- IModbusMasterWord16 : IModbusMaster
-- RegisterMultiplier (Get)   = (This is set to 1 in the 16bit implementation and 2 in the 32bit Implementation)
-- Short [Int16] (Read/Write)
-- Char (Read/Write)
- IModbusMasterWord32 : IModbusMasterWord16
-- Int [Int32] (Read/Write)
-- UInt [Uint32] (Read/Write)
-- Float (Read/Write)

There is a static function list you pass to the constructor for Endian/WordSwapping dependant on device. 
I have generic implementations of both Interfaces, 
- ModbusMasterWord16 : IModbusMasterWord16
- ModbusMasterWord32 : ModbusMasterWord16 : IModbusMasterWord32

It's heavily basted on the work here: https://github.com/Maxwe11/Maxwe11.Modbus, but i had to rewrite to be more generic, because trio can't stick to one encoding for all parameter data types, (Ascii: BigEndian), (Float: BigEndian 16bit word swapped [0A0B0C0D => 0C0D0A0B])

I'm not sure if i should submit to NModbus or a seperate project as it will need to be made even more generic for wide space use, where i expect you would have a Device specific implementation of the underlying functions. This is because Trio and Schneider appear to use mixed word sizes base on address range, and mixed Endian based on DataType.

When I finish I'll work out how to share, and where to put it.

cragdor, 2018-03-21:
Hi All, 

I have a completed the work I needed to do to support 32/64bit byte sizes. Big/Little Endian and word swapping. These are all under NModbus.Extensions.Functions.{RegisterFuctions.cs|Endian.cs}, in addition there is a NModbus.Extensions.ModbusMasterEnhanced.cs. 
It's constructor takes: 
- IModbusMaster master, 
- uint wordSize, 
- Func<byte[], byte[]> endian,
- bool wordSwap=false   // Default value

Methods: (32bit, but 64bit could easily be added)
- public char[] ReadCharHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
- public ushort[] ReadUshortHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
- public short[] ReadShortHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
- public int[] ReadIntHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
- public float[] ReadFloatHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
- public void WriteCharHoldingRegisters(byte slaveAddress, ushort startAddress, char[] data)
- public void WriteUshortHoldingRegisters(byte slaveAddress, ushort startAddress, ushort[] data)
- public void WriteShortHoldingRegisters(byte slaveAddress, ushort startAddress, short[] data)
- public void WriteIntHoldingRegisters(byte slaveAddress, ushort startAddress, int[] data)
- public void WriteUIntHoldingRegisters(byte slaveAddress, ushort startAddress, uint[] data)
- public void WriteFloatHoldingRegisters(byte slaveAddress, ushort startAddress, float[] data)

cragdor, 2018-03-21:
I have also created an additional project for my needs called : 
NModbus.Hardware
Containing: 
NModbus.Hardware.Schnieder.Servo.ILA2T.cs
NModbus.Hardware.Schnieder.Servo.ILA2T.Control.cs
NModbus.Hardware.Schnieder.Servo.ILA2T.Status.cs
NModbus.Hardware.Schnieder.Servo.ILA2T.Config.cs
NModbus.Hardware.Schnieder.Servo.ILA2T.AddressMap.cs

cragdor, 2018-03-21:
Is this something i should push back for review, would it be added to the project?
If the answer to the above is yes, how? Should i fork the project on GitHub, push the changes to my copy then offer a pull back to yours? I'm using Stash/BitBucket at home/work.

cragdor, 2018-03-21:
@rquackenbush - Folked and added code for 32bit, and raised pull request.
Currently working on implementing control code for Schneider Servo motors under a project called NModbus.Hardware, i will commit to my local git server, but can push the code back if people think it is of use, to have a Hardware library people can use.

cragdor, 2018-03-26:
Ok so looks like this is been merged in and is in the 3.0.52.
I'm going to close this as I've tested the code against Trio and Schneider.

closed 2018-03-26 by cragdor