NModbus/NModbus #20: ReadWriteMultipleRegisters datapacket incorrect

artkat, 2018-05-07:
The ReadWriteMultipleRegisters (Function code 23) only sends the address, function code and checksum. The data etc is missing.

Example:
master.ReadWriteMultipleRegistersAsync(0x99, 0x11, 3, 0x22, new ushort[] { 0xaabb, 0xccdd });
only sends (ASCII)  :991750

Fuchks, 2018-06-07:
Can confirm this behaviour, on TCP it works just fine (Tested with [Ananas](http://www.tuomio.fi/ananas/)).

As I do not have any ascii client at hand, I tested it with a fake implementation of SerialPort and SerialPortAdapter that gives access to the stream that would be sent over serial line.

ReadWriteMultipleRegisters does an aggregation of ReadHoldingRegisters and WriteMultipleRegisters in code for the request, both individual perform as expected.

I would assume that the problem is located in ReadWriteMultipleRegistersRequest at the part of aggregation, but as debugging of nuget packages is a pain, I currently cannot dive deeper into it.

On request I would provide the fake implementation of SerialPort and SerialPortAdapter, but they are quick & dirty ;-)

Fuchks, 2018-06-09:
@rquackenbush:
Here is a quick hack for the issue (imo the real issue lies deeper in concept...).
The IP-Master did not suffer from this issue, as it implements complete different BuildMessageFrame in ModbusIpTransport (not shure if that implementation fits the concept...)

Diff for ReadWriteMultipleRegistersRequest:
```
 using System;
 using System.IO;
+using System.Linq;
 using NModbus.Data;
+using NModbus.Unme.Common;
 
 namespace NModbus.Message
 {
-    internal class ReadWriteMultipleRegistersRequest : AbstractModbusMessage, IModbusRequest
+    internal class ReadWriteMultipleRegistersRequest : AbstractModbusMessageWithData<RegisterCollection>, IModbusRequest
     {
         private ReadHoldingInputRegistersRequest _readRequest;
         private WriteMultipleRegistersRequest _writeRequest;
 
         public ReadWriteMultipleRegistersRequest()
         {
         }
 
         public ReadWriteMultipleRegistersRequest(
             byte slaveAddress,
             ushort startReadAddress,
             ushort numberOfPointsToRead,
             ushort startWriteAddress,
             RegisterCollection writeData)
             : base(slaveAddress, ModbusFunctionCodes.ReadWriteMultipleRegisters)
         {
             _readRequest = new ReadHoldingInputRegistersRequest(
                 ModbusFunctionCodes.ReadHoldingRegisters,
                 slaveAddress,
                 startReadAddress,
                 numberOfPointsToRead);
 
             _writeRequest = new WriteMultipleRegistersRequest(
                 slaveAddress,
                 startWriteAddress,
                 writeData);
+
+            // TODO: ugly hack for all ModbusSerialTransport-inheritances (ModbusIpTransport would not need this, as it implements complete different BuildMessageFrame)
+
+            // fake ByteCount, Data can hold only even number of bytes
+            ByteCount = (ProtocolDataUnit[1]);
+
+            // fake Data, as this modbusmessage does not fit ModbusMessageImpl
+            Data = new RegisterCollection(ProtocolDataUnit.Slice(2, ProtocolDataUnit.Length - 2).ToArray());
+        }
+
+        public byte ByteCount
+        {
+            get => MessageImpl.ByteCount.Value;
+            set => MessageImpl.ByteCount = value;
         }
```

closed 2018-06-27 by rquackenbush