NModbus/NModbus #23: SerialPortStreamAdapter must be reference 2.1.2 version

pigwing, 2018-06-13:
hi i am update NModbus.SerialPortStream 3.0.55 and SerialPortStream 2.1.4 bug SerialPortStreamAdapter must be reference 2.1.2 version

AJamesEdwards, 2018-07-06:
var modbusFactory = new ModbusFactory();

            string serialPortName = SerialPortStream.GetPortNames().First();
            using (var comport = new SerialPortStream(serialPortName, 19200, 8, Parity.Even, StopBits.One))
            {
                comport.Open();

                using (var adapter = new SerialPortStreamAdapter(comport)) //<== see error message below
                {

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0012	The type 'SerialPortStream' is defined in an assembly that is not referenced. You must add a reference to assembly 'RJCP.SerialPortStream, Version=2.1.2.0, Culture=neutral,

pigwing, 2018-07-08:
i found that use serialPortStream 2.1.4 will let the cpu 100%....so...use the 2.1.2 is correct

AJamesEdwards, 2018-07-10:
Thanks that Worked !!!

closed 2018-09-11 by rquackenbush