NModbus/NModbus #28: How can I read and write negative numbers?

xzoth, 2018-08-15:
I'm a novice, the first time using Modbus protocol for communication, after a simple attempt, I found that nmodbus uses USHORT data type for reading and writing, then how to read and write negative numbers? Thank you

MPapst, 2018-08-15:
Hi,
you will need to convert the input format (e.g. signed integer or even floating numbers) into its bit representation.
The reason why an ushort is used for modbus is that registers are organized in 16 Bits.

```C#
int inputValue = -15;
byte[] inputBytes = BitConverter.GetBytes(inputValue); 
// inputBytes now contains 4(!) bytes with the bit representation... 
// now you will need to convert them to ushort --> 
// two bytes = 1 ushort, the ordering depends on your system architecture.. google for Endianess

// one way is
ushort lower = inputBytes[0] + (inputBytes[1] << 8);
ushort upper = inputBytes[2] + (inputBytes[3] << 8);
```

Please note that the code is completely untested and does not have any security mechanisms like try/catch or null checks.

xzoth, 2018-08-15:
Thank you very much. 

I found other ways from google:
`ushrotValue=intValue-((intValue>32767)?65536:0);`
or
`ushortValue=BitConverter.ToUInt16(BitConverter.GetBytes(intValue),0);`

closed 2018-09-11 by rquackenbush