NModbus/NModbus #29: NModbus.Hardware

cragdor, 2018-09-06:
Hi All, 

After the adding the extension functions for supporting short/uint/int/float I have refactored some of my code to use a couple of generic device wrappers. One for Trio Motion Controllers and one for Schneider Electronics motor controllers.  
I have placed these two wrappers into a library called 'NModbus.Hardware', as examples of the way to talk to devices with 16/32 bit words. 
Is this of use to the project? It was my hope that the library could be committed back and that other examples of communicating with hardware could be added?

[image]

Trio motion controllers are programmed by the end user so it has a basic wrapper. The device can be configured to be UShort/Integer/Float so there is a wrapper for each.

Schneider servo motors run a program written by Schneider that you interface with over Modbus. The parameters of the program can be ushort/short/uint/int/float, so there is a basic wrapper for those wanting to make their own implementation. 
I have included the documentation and basic implementation of a Lexium ILA2T device, to show how to interface.

@rquackenbush should I create a pull request? Is this useful? 

Craig

rquackenbush, 2018-09-11:
Hi @cragdor. I do some similar operations (read / write 32 bit floating point values, 32 bit signed, unsigned integers, etc) in my applications. I can see where a utility that works with registry ranges of data could be useful. Maybe we could come up with something that reads from / writes to `ushort` arrays like this:

```c#
public static void WriteFloat32(this ushort[] registers, int startIndex, float value) 
{
}

public static float ReadFloat32(this ushort[] registers, int startIndex)
{
}
```

This way everybody has access to a centralized implementation for this sort of operation, but it's not tied directly to reading / writing Modbus data. Is this close to what you have in mind?

cragdor, 2018-09-12:
Hi @rquackenbush ,

The operations extensions I added back last time make it easier to talk to 32/64bit operators. Effectively now you just: 

```cs
public float[] ReadFloatHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
            => RegisterFunctions.ByteValueArraysToFloats(
                RegisterFunctions.ReadRegisters(slaveAddress, startAddress, numberOfPoints, this.master, this.wordSize, this.endian, this.wordSwap), this.frontPadding);
// this.frontPadding used by TrioMotionController  ushort on int config 0xFE0A -> 0x0000FE0A

public void WriteFloatHoldingRegisters(byte slaveAddress, ushort startAddress, float[] data)
             => RegisterFunctions.WriteRegistersFunc(
                slaveAddress,
                startAddress,
                RegisterFunctions.FloatToByteValueArrays(data, this.wordSize, this.frontPadding),
                this.master,
                this.wordSize, //16/32/64
                this.endian, // Functions.Endian.BigEndian/LittleEndian
                this.wordSwap); //Used for some pieced of hardware ABCD -> CDAB sometimes know as mirroring bytes.
```

This library was just a collection of examples of how to talk to the hardware. The trio class just has a class implementation for ushort/integer/float, which is how the device can be configured, the Endian, wordsizes, wordswaps are consts at the begin of each class. 

However the Schneider electric implementations is more complex, different registers have different data types. So the wrapper class has Read/Write<ushort/short/uint/int/float>Registers.
There is a skeleton implementation of the Lexium Commands to interface with the device. 

Both are implementations I use regularly and I wanted to share as the documentation supplied for both devices are miss leading and in some cases just wrong (Wireshark to the rescue). 

The end goal I suppose was to have a collection of skeleton libraries for interfacing with different hardware, or at least a collection of working examples people could follow.