NModbus/NModbus #31: Shared DataStore between multiple Slaves

ARaddi99, 2018-09-18:
Hello,
I have 2 PLCs and I have to write in the same register two different values, unfortunately as I change register number 100 in PLC1, instantly changes PLC2's register 100 with the same value of PLC1's. I could see that DataStore1 contains that value, while DataStore2 doesn't, it is filled with 0.
How am I supposed to achieve what I need?

rquackenbush, 2018-09-18:
Hi @ARaddi99 - I'm not quite following what you're running into. Are you coding a Modbus Master to run against PLCs that are Modbus slaves?

AJamesEdwards, 2018-09-18:
Do you have both PLC's set to the same RTU address???

ARaddi99, 2018-09-18:
@rquackenbush I’m coding a Modbus slave to run against PLCs that are Masters. I’m connecting them with TCP not serial port. Slaves point at the same IP address, the slave one. So far my application is just listening and catching the WrittenCompleted event.

ARaddi99, 2018-09-18:
@AJamesEdwards I’m not using RTU connection

MPapst, 2018-09-18:
When both PLC are connecting to your server (slave) and using the same operations to write their data, then it is as expected that the values are not differentiated by your PLC inside the server.
In general the modbus slave accepts serveral masters to write into its registers, but is not creating different registers per client. The easiest way to achieve this would be to use different unique ID's (I have not checked if NModbus supports this, but I assume this). 
Nevertheless - in my opinion - you are doing it the wrong way. The PLC's should be your slave and your application should use two masters to read their data.

ARaddi99, 2018-09-18:
@MPapst Using my application as Master, is there an event that catch the change of a register? Because with another library I had to use a timer to read the register's content and if it wasn't empty I would do some operations, so it was easier with change event that I found in this library. So my question is: does it exist even as Master?

AJamesEdwards, 2018-09-18:
OK, not using a serial RTU connection.  But just to be complete do both
PLC's have the IP Address?

ARaddi99, 2018-09-18:
@AJamesEdwards Yes, they both have an IP Address, that is obviously different, and both of them point to the same Slave IP Address (my application).

MPapst, 2018-09-19:
@ARaddi99 no, this is not how modbus works. Modbus works with a top-down model, having the highest application/controller (like a scada system for example) as multiple master, adressing the controllers below by polling their data.
If you want to achieve that the underlying controller can push its data, you are in need of a protocoll that supports a publish/subscribe model like MQTT does.
If you want to achieve this via modbus the traditional approach would be to use one slave per PLC (you see why I think this is the wrong way?). With NModbus an its DataStores it **would** be possible to write a DataStore that has different data storage per connected client - but I think this is not easy handable if you want to process the data within you application and you would be doing something that is not as expected for modbus (which is always critical if someone else is working with your application/system).

Actually if you want to use modbus I would suggest do a cyclic polling of the data using a scheduling library (Quartz.NET did a good job for me with a really small jitter and very accurate triggers even in subsecond schedules).

In simple implementations (e.g. the OSCAT Modbus implementation for structured text or simple protocol adapters [e.g. Modbus/TCP to RTU adapter]) in a modbus slave data is written/read from/into a specified range of RAM and other parts of the application are just reading this RAM.

AJamesEdwards, 2018-09-19:
Agreed. Modbus is a simple Master / Slave protocol.  Slave devices only
speak when polled by a master.
In a situation with one slave and several masters it ends up as a race
condition where the last master to write a value wins.

ARaddi99, 2018-09-20:
Thank you all for your answers. I’m going to change my application and PLCs making ‘n’ Slaves and one Master

closed 2018-09-20 by ARaddi99