NModbus/NModbus #34: Some modbus support is not a very standard device, there is a problem

jackswei, 2018-11-24:
I sent id 1 address 72 get 1 it's no data
I sent Id 1 address 72 get 2 it's have data, but only one data , so in NModbus happen wrong, but others modbus softwere is ok,i think it no strict. 

how can slove in  very standard device problem?

MPapst, 2018-11-24:
Hi,
this is a very common issue with modbus and different device manufacturers (or used libraries on device side). 
The background is that some libraries are starting to count at 0 and some at 1.

This is nothing that is special with the nmodbus lib. The easiest way to figure out how you need to create your modbus request is to create a wireshark (for ModbusTCP) or a RS485 Capture (for RTU) and compare that with the two tools/libraries you are using on PC side.

AJamesEdwards, 2018-11-25:
Or poll an address ( eg. FC4 register # 30) with a know value in it and add
/ subtract one from the address until you receive the correct value.

Note: NModbus uses the FC and the register index into an array. So don't
try to poll FC4 address 30,030. Instead poll FC4 address 30.

jackswei, 2018-12-10:
I TRY USE RMMS OF MODBUS TOOL  ,IT IS OK. I THING GET 2 DATE ,BUT ONLY SNED 1 DATE,SO NMODBUS    'S DROP THIS.
[image]
[image]
[image]

closed 2020-04-15 by rquackenbush