NModbus/NModbus #39: Support for Function Code 20 and 21 (0x14 and 0x15)

wildbiotiger, 2019-01-14:
Is there any plan to support the File Record R/W Modbus Function Codes?

rquackenbush, 2019-01-15:
There are no current plans to support as I don’t have any equipment to test
with that supports it. Would you be willing to provide a pull request with
an implementation?

wildbiotiger, 2019-01-16:
Possibly, I'm at a loss for reliable hardware as well. If there's a point where I can take this on, I'll let you know. Thanks for responding, though.

smkanadl, 2020-03-09:
I created a PR #58 to add first basic support of function code 21 to modbus master. I am not a pro on modbus protocol stuff, so please be gentle :-)

smkanadl, 2020-04-15:
@rquackenbush It's probably to early to close this issue because function code 20 is still missing. Or will you just wait for someone to create a new issue. But having this "just" closed is probably misleading.

rquackenbush, 2020-04-15:
@smkanadl good point. Will leave this open.

pgolinski, 2022-01-12:
Hello. I'm working on code 20 support. Just for your confidence - tests are failing on current HEAD. Example:
Test expects ModbusMaster.WriteFileRecord to throw ArgumentException
https://github.com/NModbus/NModbus/blob/develop/NModbus.UnitTests/Device/ModbusMasterFixture.cs#L78
but ArgumentNullException is thrown
https://github.com/NModbus/NModbus/blob/develop/NModbus/Device/ModbusMaster.cs#L414

I will fix that in my incoming PR

pgolinski, 2022-01-13:
I have the implementation ready, but I must wait for a device on which I will have possibility to test it.