NModbus/NModbus #42: DataStore Pointsource to spec mapping

WattsC-90, 2019-02-11:
Hopefully someone can clarify this for me, I hope that the resolution of this query may also improve the naming of some properties too..

The modbus spec defines the following function codes:

- `Read Coil`
- `Read Discrete Input`
- `Write Single Coil`
- `Write Multiple Coils`

From this, the modbus spec defines their behaviours/types:

- Coils are 1 bit registers, and can be read or written.
- Discrete Inputs are 1 bit registers, read only.

In the `ISlaveDataStore` code, there are 4 `IPointSource` properties, two are bool (and related here):

- `CoilDiscretes`
- `CoilInputs`

Can someone please tell me which these two properties map to? It is entirely unclear to me from the documentation and the code and I don't want to write a modbus master to have to exercise the interface to figure out which one is which.

The function codes `ReadCoil`, `WriteSingleCoil` and `WriteMultipleCoils` map to which `IPointSource<bool>`? `CoilDiscretes` or `CoilInputs`?

Similarly, which property does `ReadDiscreteInput` map to?

It is of course, entirely possible that I have misunderstood the point of the datastore, so I apologise if that is the case, but I am so totally confused by this point I need someone to point me in the right direction :D

rquackenbush, 2019-02-13:
I didn't use very good names for those `ISlaveDataStore` members. They should probably have been:

- `Coils` (e.g. `ReadCoil`, `WriteSingleCoil`, `WriteMultipleCoils`)
- `Inputs` (e.g. `ReadDiscreteInput`)

Apologies for the unclear names. I might add the correct names and obsolete the old ones at some point.

WattsC-90, 2019-02-18:
thanks for the input @rquackenbush ! So which ones map to which as it currently stands?

rquackenbush, 2020-04-15:
@WattsC-90 - apologies for the incredibly long delay.

You've probably already sorted this, but:

|Spec|NModbus|
|---|---|
|Discrete | CoilDiscretes |
|Coils | CoilInputs |
|a |b |

closed 2020-04-15 by rquackenbush

WattsC-90, 2020-09-23:
Again hit this problem.. more confusing this time.. I have an app for simulation, running the slave/server end.. with a webapi on top.. it writes to "`CoilInput`" datastore and "`CoilDiscrete`" datastore for `Coils` and `DiscreteInputs` respectively.. the master/client is inverse.. (logically speaking based on naming): "`ReadInputs`" and "`ReadCoils`" is actually for Coils and Inputs respectively..

This should just be changed in 3.x to become `ReadCoil` or `ReadDiscreteInput` for master/client, and `Coil` and `DiscreteInput` datastores for the slave/server end. Totally barmy! :D

rquackenbush, 2020-09-23:
Agreed. The naming got out of hand. We'll sort it in 3.x

WattsC-90, 2020-10-05:
@rquackenbush  when is 3.x going to drop (ish?) im falling foul of the naming again.. driving me mad now!

WattsC-90, 2020-10-05:
seems like `writesinglecoil` comes out on `coildiscretes` datastore. which seems opposed to what i understand of discrete inputs (non-writeable).. and goes against the naming above.