NModbus/NModbus #43: Reading many registers at once

tomsimmons78, 2019-03-27:
I'm looking to read several registers at a single point in time.

I was hoping I could do a ReadInputRegisters (0,0,x), getting all the registers in one go, then pick out the ones I want.  My first register is at address 0, another is at 3008.  The number of registers you can read (x above) seems to be 125 (entered as 124) which means I can't read all of them at once.

Why is there a limit on the number of registers you can read?

Has anyone done something similar to what I'm looking at?

Regards

Tom

MPapst, 2019-03-27:
The limitation comes from the Modbus protocol itself. It limits the read of Input and Holding registers to 125.
The Modbus protocol functions that are use below ReadInputRegisters can read this 125 registers at once. But due to another limitation of the Modbus protocol these must be subsequent:
So if you start reading at address 0, you can read all registers from 0 up to registers 124.

Please note that depending on the device to which you are connected you may or may not read over non existing registers.

tomsimmons78, 2019-03-27:
Ahh, OK.

So it would seem that I would need to use threads, 1 for each register, and sync them to all read at the same time.

Tom

rquackenbush, 2019-03-27:
@tomsimmons78 - that approach won't work with serial devices, and might overwhelm resource constrained IP based devices. 

Here's an approach I use to query more than 125 data points for a given range:

```c#
    public static class ModbusExtensions
    {
        public static async Task<ushort[]> ReadHoldingRegistersChunked(this IModbusMaster master, byte slaveAddress, ushort startAddress, int numberOfPoints, int chunkSize)
        {
            return await ReadChunkedAsync(startAddress, numberOfPoints, chunkSize,  (a, n) => master.ReadHoldingRegistersAsync(slaveAddress, a, n));
        }

        public static async Task<ushort[]> ReadInputRegistersChunked(this IModbusMaster master, byte slaveAddress, ushort startAddress, int numberOfPoints, int chunkSize)
        {
            return await ReadChunkedAsync(startAddress, numberOfPoints, chunkSize, (a, n) => master.ReadInputRegistersAsync(slaveAddress, a, n));
        }

        public static async Task<bool[]> ReadInputsChunked(this IModbusMaster master, byte slaveAddress, ushort startAddress, int numberOfPoints, int chunkSize)
        {
            return await ReadChunkedAsync(startAddress, numberOfPoints, chunkSize, (a, n) => master.ReadInputsAsync(slaveAddress, a, n));
        }

        public static async Task<bool[]> ReadCoilsChunked(this IModbusMaster master, byte slaveAddress, ushort startAddress, int numberOfPoints, int chunkSize)
        {
            return await ReadChunkedAsync(startAddress, numberOfPoints, chunkSize, (a, n) => master.ReadCoilsAsync(slaveAddress, a, n));
        }

        // It turns out that we need to read ranges that are longer than the Modbus maximum (125). So, to support that, we can read them by "chunks". This will also 
        // help support resource constrained Modbus devices.
        private static async Task<TValue[]> ReadChunkedAsync<TValue>(ushort startAddress, int numberOfPoints, int chunkSize, Func<ushort, ushort, Task<TValue[]>> func)
        {
            var result = new List<TValue>(numberOfPoints);

            int remaining = numberOfPoints;
            int soFar = 0;

            while (remaining > 0)
            {
                int thisPass = remaining;

                if (thisPass > chunkSize)
                {
                    thisPass = chunkSize;
                }

                var registers = await func((ushort)(startAddress + soFar), (ushort)thisPass);

                foreach (var register in registers)
                {
                    result.Add(register);
                }

                remaining -= thisPass;
                soFar += thisPass;
            }

            return result.ToArray();
        }
    }
```

tomsimmons78, 2019-03-27:
@rquackenbush, thanks for the reply.

I already suspected the multiple threads wouldn't work.

I've not had the chance to read your reply properly, I'm confined to viewing by a mobile at this precise moment. However, if it is not possible to receive more than 125 registers back in a single request I can't see how it would be possible to read any more than 125 at the exact same point in time, meaning reading the next 125 however quickly would be a fraction of a second later.

As I say, your solution may well handle, this I'll review as soon as I can.

Tom

rquackenbush, 2019-03-27:
@tomsimmons78 - yes - that's a downside to the batch reading approach. You have to be careful that you don't split across a set of registers that requires an atomic read (e.g. a 32 bit integer or float value). If you have this scenario, you'll have to be more explicit as to how you break out your reads.

tomsimmons78, 2019-03-27:
@rquackenbush my problem isn't splitting two registers that should be read atomically. The client wants 6 sensors which are 2 registers a piece read at a single point in time at 32Hz.

The IS1+ from Stahl, according to Stahl, doesn't allow for the registers used to be pre-assigned, so the six pairs could appear anywhere rather than consecutively.

If I cant read all possible registers atomically, then I can't see how I can capture all the required data at the single point in time.

Tom

MPapst, 2019-03-28:
32Hz means ever 31.25 ms a read to all datapoints.
I do not think that .net or .net core is the right solution for this, as the Garbage Collection may or may not trigger and cost some of the time you need. Even the Windows time resolution is (I think to remember) something aroung 12ms. So it may not even be accurate.

You would better go with something like Codesys (Soft Real Time PLC) or an embedded C device to handle the requests and use a different protocol for your upstream (e.g. a Push Protocol like MQTT).
The embedded or PLC world thinks in such cycles and you even could use multiple clients to try to gather those six data points at the same time.
If you need it atomic, you woun't be able to use this setup (I do not think Modbus is the right solution for atomic reads - only if you control both sides, but not in a standard scenario).

closed 2020-04-15 by rquackenbush