NModbus/NModbus #44: Where is NModbus.chm

schungx, 2019-03-31:
Please tell me where is the elusive `NModbus.chm` which supposedly holds the documentation?

AlexanderNitsch, 2020-01-28:
I found it in the previous version, that this one was forked from: https://github.com/NModbus4/NModbus4/blob/master/NModbus.chm

But it would be nice, if it were in the repo.

AlexanderNitsch, 2020-01-28:
Does not seem to fit the new implementation fully, though... But its a starting point

rquackenbush, 2020-01-28:
Here is the updated documentation:

https://nmodbus.github.io/api/NModbus.html

kjetilgloppen, 2020-04-03:
FYI: It was removed in commit bec1a51346df0f. 
Not sure if it was intentional or not, but the commit message doesn't mention this change.

rquackenbush, 2020-04-15:
Yes - it was removed on purpose. chm is a pretty outdated format. I've updated the README.md to point to the latest.

closed 2020-04-15 by rquackenbush