NModbus/NModbus #45: Plans for integration of new Modbus Security Specification

MPapst, 2019-04-18:
At 29th October 2018 the Modbus Organization has released the new Protocol Improvement with TLS Encryption and Authentication.
Are there any plans to implement this?

Press release: http://www.modbus.org/docs/Modbus-SecurityPR-10-2018.pdf
Security Spec can be found on the Specs Page: http://modbus.org/specs.php

rquackenbush, 2020-04-15:
This is on the roadmap (no timeline yet). I'm curious to see what vendors will adopt the standard.

MPapst, 2020-04-15:
Te Company I work for is waiting also if someone else wants to implement this.

devsbazefield, 2023-05-31:
The company I am working for are considering sponsoring this feature since we are getting requests from customers regarding this. Please contact me on tollef.slaathaug@bazefield.com for more details.

rquackenbush, 2023-08-21:
An alpha release of this feature is available in the 4.0.0-alpha008 version of NModbus.

Note:
- This is a complete rewrite of NModbus from the ground up. It is made of breaking changes.
- Only UDP/TCP transports are available at this time