NModbus/NModbus #51: Slave not responding

ruciuPL, 2019-07-30:
I am using NModbus to connect PC (master) to a PLC (slave).
Everything runs perfect until PLC goes to STOP or ERROR - the connection remains, however the Modbus Slave is not responding. 
I am using ReadMultipleRegisters function. The telegram is sent to PLC but there is no response and program is waiting for response. Is there a response timeout method to use? Should I do it any other way?

Fuchks, 2019-07-30:
Yes, you have to set the timeouts for e.g. tcpclient - this is out of scope of NModbus lib.

ruciuPL, 2019-07-31:
I am thinking about using a timer to check if "response" is != NULL. Should it work? How should I perform a "break" istruction for a higher-level function (I am out of date in c#)

Fuchks, 2019-07-31:
There are built-in functions for that.
For TcpClient set ReceiveTimeout and SendTimeout.
For SerialPort set ReceiveTimeout.

ruciuPL, 2019-07-31:
Thanks, I will try it ASAP

ruciuPL, 2019-07-31:
Another issue... If a PLC goes to STOP and return to RUN when I send data to it the "slace busy" exception occurs and blocks the whole program (in this case only Modbus in PLC is stopped, the TCP/IP layer is running). I have no idea how to manage this.

closed 2019-11-21 by jesbrd