NModbus/NModbus #52: ModbusTcpSlaveNetwork.ListenAsync will not cancel

edwwsw, 2019-09-09:
ListenAsync will not honoring cancel request.  I believe the issue is the inner loop of the ListenAsync  is "stuck" waiting on a accept request so the while loop can not check the cancel request. 

            while (!cancellationToken.IsCancellationRequested)
            {
=>              TcpClient client = await Server.AcceptTcpClientAsync().ConfigureAwait(false);
                var masterConnection = new ModbusMasterTcpConnection(client, this, ModbusFactory, Logger);
                masterConnection.ModbusMasterTcpConnectionClosed += OnMasterConnectionClosedHandler;
                _masters.TryAdd(client.Client.RemoteEndPoint.ToString(), masterConnection);
            }

edwwsw, 2019-09-09:
Just verified that I can get the Tcp Slave to cancel if I connect a master  after making a cancel request. 
             
                CancelToken?.Cancel();

                // Hack to get out of task loop.  Connect a client. 
                if (ModbusTask != null)
                {
                    var factory = new ModbusFactory();
                    var TCPClient = new TcpClient("127.0.0.1", 502);
                    var Modbus = factory.CreateMaster(TCPClient);
                    System.Threading.Thread.Sleep(1000);
                }
      

                if (ModbusTask?.Wait(10_000) == false)
                {
                    PLogger.GetInstance().Error("Failed to close down Modbus Slave");
                    ret = false;
                }

                TCPListener?.Stop();
                NetworkSlave?.Dispose();

kjetilgloppen, 2020-04-17:
I noticed the same issue. The `ListenAsync()` call will not return until a Master has connected.

closed 2020-07-14 by rquackenbush