NModbus/NModbus #55: Dependency on older Version of SerialPortStream

hayozsoft, 2020-01-15:
I'm using NModbus.SerialPortStream in my environment as i'm using .NET-Core.

However, the Nuget-Package NModbus.SerialPortStream seems to have a dependency on SerialPortStream 2.1.2.
In my project i need the newest Version of SerialPortStream (Version 2.2.0).

I have to use SerialPortStream outside of the NModbus-Lib to access more devices over RS232 and RS485.

Is there a way to change this dependency that it's not installing and using an older version of SerialPortStream?

Thx!

rquackenbush, 2020-01-17:
By any chance can you upgrade to .NET Core 3.1? SerialPort is now supported
(finally) on Linux. That way you don't have to worry about SerialPortStream
at all (at least for NModbus).

rquackenbush, 2020-04-15:
I've committed this. It will be available in the next release.

closed 2020-04-15 by rquackenbush