NModbus/NModbus #57: Exception thrown: 'System.NotImplementedException' in NModbus.Serial.UWP.dll

BullT, 2020-03-05:
Hi,

I'm trying to tun a NModBus on a Raspberry Pi using UWP.

I can comunícate with ModBus modules but during the programm running in the Output windows (Visual Studio) appear every time that I refresh Modbus the comment: Exception thrown: 'System.NotImplementedException' in NModbus.Serial.UWP.dll.

My code is this:

`private async void MainPage_Loaded(object sender, RoutedEventArgs e)
        {
            try
            {
                using (SerialDevice serialPort = await SerialDevice.FromIdAsync("\\\\?\\USB#VID_1A86&PID_7523#6&CF77267&0&1#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"))
                {
                    if (serialPort == null) return;

                    // configure serial port
                    serialPort.WriteTimeout = TimeSpan.FromMilliseconds(1000);
                    serialPort.ReadTimeout = TimeSpan.FromMilliseconds(1000);
                    serialPort.BaudRate = 9600;
                    serialPort.Parity = SerialParity.None;
                    serialPort.StopBits = SerialStopBitCount.One;
                    serialPort.DataBits = 8;
                    serialPort.Handshake = SerialHandshake.None;

                    var factory = new ModbusFactory();
                    var adapter = new SerialDeviceAdapter(serialPort);
                    IModbusMaster ModBusMaster = factory.CreateRtuMaster(adapter);

                    BoardsDefinitions();
                    InputsDefinitions();
                    OutputsDefinitions();
                    LightsDefinitions();

                    await Task.Delay(TimeSpan.FromMilliseconds(2000));
                    do
                    {
                        foreach (var light in lights)
                        {
                            foreach (var input in light.Input)
                            {
                                var inputStatus = ModBusMaster.ReadInputs(input.BoardName.StationId, 0, input.BoardName.NumberOfInputsOutputs);
                                if (inputStatus[input.InputOutputNumber] != input.InputOutputStatusOld)
                                {
                                    foreach (var output in light.Output)
                                    {
                                        ModBusMaster.WriteSingleCoil(output.BoardName.StationId, output.InputOutputNumber, !output.InputOutputStatusOld);
                                        output.InputOutputStatusOld = !output.InputOutputStatusOld;
                                    }
                                    input.InputOutputStatusOld = !input.InputOutputStatusOld;
                                    break;
                                }
                            }
                        }
                        await Task.Delay(TimeSpan.FromMilliseconds(50));
                    } while (true);
                }
            }
            catch (Exception ex)
            {
                throw;
            }
        }`

Can you help me to solve this exception?

Thanks,
XCO

rquackenbush, 2020-04-15:
@BullT - do you know what specific line of the above code leads to the exception getting logged in the output window?

wmjb, 2021-01-10:
> @BullT - do you know what specific line of the above code leads to the exception getting logged in the output window?

im in same boat. leaking badly.

traced it to Write(message); in ModbusTransport.cs

public virtual T UnicastMessage<T>(IModbusMessage message)
            where T : IModbusMessage, new()
        {
            IModbusMessage response = null;
            int attempt = 1;
            bool success = false;        
            do
            {
                try
                {
                    lock (_syncLock)
                    {
                        Write(message);

wmjb, 2021-01-13:
> @BullT - do you know what specific line of the above code leads to the exception getting logged in the output window?

narrowed it down,

SerialDeviceAdapter

       public void DiscardInBuffer() 
        {
           Task.Run(async () => await _serialDevice.OutputStream.FlushAsync());
        }

BullT, 2021-01-14:
Hi,

Due to several concerns I decided to move to Python.
Sorry I can't help you.