NModbus/NModbus #61: Bug in ModbusSerialSlaveNetwork

MrAliev, 2020-05-15:
Я использую NModbus для эмуляции нескольких Mobdus-RTU Slave устройств.
Когда Modbus-Master делает запросы к Modbus-Slave с адресом отсутствующего на шине устройства, то я получаю ложные ошибки в логе.

На этом скрине добавлены все 6 ведомых, к которым обращается ведущее устройство:

[image]

Здесь, я убрал устройство Id-6, и получаю ошибки:

[image]

В моем случае, ко всем ведомым устройствам отправляются абсолютно одинаковые запросы (нет не поддерживаемых функций и т.д.).
Ошибки возникают только если приходит запрос к ведомому устройству которого нет.

MrAliev, 2020-05-16:
Кажется я нашел проблему:
NMonbusRtuTransport.cs
public override void IgnoreResponse()   - в этом методе производится еще одно считывание пакета из буфера (для вывода содержимого в лог), но у меня там уже находятся другие данные...
Я не настолько сильно разбираюсь в программировании и в ModBus, чтобы грамотно пофиксить этот баг, так что я просто отключил "лишнее" считывание, чтобы избавиться от ошибок.
Для моего случая это работает.
[image]

Теперь все работает правильно, а в лог выводится сообщение об игнорировании запроса к другому устройству без вывода содержимого самого пакета.
[image]

chenjing1294, 2023-08-05:
I don't understand what the ReadResponse method does, why is it reading some special length bytes? Can anyone explain this?