NModbus/NModbus #62: Encountered an Issue with reconnecting to the Service

CLST-Isopp, 2020-05-25:
I was wondering if reconnection with the NModus Service is supported as I tried to reconnect but the system keeps throwing me exception errors regarding the socket. 

Exception thrown: 'System.Net.Sockets.SocketException' in System.dll
Exception thrown: 'System.IO.IOException' in System.dll
Exception thrown: 'System.IO.IOException' in NModbus.dll

Maybe I'm implementing it wrong?

MPapst, 2020-05-25:
- Can you open a plain socket to your target system on the specified port?
- When does that happen? when connecting or during the connection?
- Can you copy the complete exception message?

CLST-Isopp, 2020-05-26:
So I might have been unclear. The connection works fine the first time. So yes i can open the socket I can open the connection and everything is working perfectly. When I pull the network cable from the SPS-Modul I'm connecting to using TCP-IP it throws this error. But when I try to reconnect, plugging the SPS-Modul back into the PC. It cannot recreated the connection and continously throws the above mentioned errors.

System.Net.Sockets.SocketException (0x80004005): No connection could be made because the target machine actively refused it 127.0.0.5:502 at System.Net.Sockets.TcpClient..ctor(String hostname, Int32 port)

        TcpClient client;
        IModbusMaster master;
        ModbusFactory factory;
        string ipAdresseSave;
        int portSave;
        public bool ModbusTcpMasterConnection(string ipAdress, int port)
        {
            
            if (client == null)
            {
                client = new TcpClient();
                ipAdresseSave = ipAdress;
                portSave = port;
            }
            client = new TcpClient(ipAdress, port);            
            factory = new ModbusFactory();
            master = factory.CreateMaster(client);
            client.ReceiveTimeout = 5000;
           
            return client.Connected;
        }

        public void ReadValues(ref byte[] data)
        {
            if (client.Connected)
            {
                ushort startAddress = 1504;
                ushort numInputs = 9;
                ushort[] values = master.ReadHoldingRegisters(1, startAddress, numInputs);                
                Buffer.BlockCopy(values, 0, data, 0, values.Length * 2);
            }            
            ModbusTcpMasterConnection(ipAdresseSave, portSave);
        }

MPapst, 2020-05-26:
Try against a Server program on a Desktop Computer.
You can use Ananas64 for example.

I‘ve seen a lot of modbus implementations, especially on PLC devices that are not clearing a socket when the client is disconnected.

Can you reconnect after a while (an hour or so)?