NModbus/NModbus #66: Suggestions wanted for naming conventions

rquackenbush, 2020-07-22:
The names `Master` and `Slave`, while coming directly from the spec https://modbus.org/specs.php are outdated.

There is a movement underway to remove these kinds of terminology from code:
https://sanfrancisco.cbslocal.com/2020/07/03/twitter-inclusive-code-language-removing-master-slave-blacklist/

Some potential replacements:

- `Leader` / `Follower` (I don't feel that this is a slam dunk)
- I like `Device` to replace `Slave`. I don't have a good corresponding name to replace `Master`.
- I feel that `Server` / `Client` is confusing in this context (where the `Slave` is technically the server, while the `Master` is the client). I feel that this is even more confusing for serial transport scenarios.

MPapst, 2020-07-22:
Listener could be a good replacement for Slave
Maybe Initiator could replace Master then.

Fuchks, 2020-07-23:
Definitely Client/Server is misleading for modbus.
Device does not describe the role, would not be my favorite.
Talker/Listener, Publisher/Subscriber and Provider/User are other common terms.
Initiator seems uncommon, but describes the role of the Master quite good.

MPapst, 2020-07-23:
Initiator comes from IPSec.

I think Provider and Consumer is common for messaging and doesnt describe the polling behavior from modbus.

MPapst, 2020-07-24:
just to add: Over the industry I am in contact with, server and client is quite common for Modbus/TCP.
And as note the initial text, the term Device maybe suitable for simple connections between a server and client, but in my terms I often have to deal with multiple stages of modbus structure (e.g. the first client offers a server for a second stage).

bdube, 2021-01-29:
Where I work, we're using `Peripheral` where `Slave` is normally used, and `Device` for a particular set of registers. So one `Peripheral`, a Schneider M172 for example, can have many `Device` objects associated with it.

I also like the suggestion of `Device` in place of `Slave`, and like to avoid client/server completely for serial RTU.

sparkchaser, 2021-02-02:
IMO, `initiator` and `target`/`endpoint` would be more consistent with how the nodes actually behave.

To minimize confusion, though, you should use the same terminology as the Modbus spec itself.  It looks like they're [using "client" and "server"](https://www.modbus.org/docs/Client-ServerPR-07-2020-final.docx.pdf).

mycroes, 2022-04-05:
I don't necessarily mind alternate terminology, but in my experience you want to align terminology with your partners. So as long as the Modbus organization refers to master and slave, I think that's the terminology you should use. If you want to change that, you should push them to make a change. And when they make the change, just refactor and do a major release with the changed terminology, perhaps keeping the old terminology in with `Obsolete` attributes for one major release cycle.

rquackenbush, 2022-09-30:
Thanks @sparkchaser for that resource. That clinches it - client server it is.

closed 2022-09-30 by rquackenbush