NModbus/NModbus #67: 3.x Roadmap

rquackenbush, 2020-07-22:
Items under consideration for version 3.x:

- [Modbus Security Specification](https://github.com/NModbus/NModbus/issues/45)
- [Naming Conventions](https://github.com/NModbus/NModbus/issues/66)
- [Fix Modbus object names](https://github.com/NModbus/NModbus/issues/42)

MPapst, 2020-07-22:
what about: Hosting a Modbus TCP or UDP listener inside kestrel using endpoints?

Ryba1986, 2021-02-13:
Please replace the "NModbus.Serial" library with the native "System.IO.Ports"

rquackenbush, 2022-09-30:
> what about: Hosting a Modbus TCP or UDP listener inside kestrel using endpoints?

@MPapst - do you have a use case for this?

rquackenbush, 2022-09-30:
FYI - I've started coding the next major version of NModbus here: https://github.com/NModbus/NModbusNext

Any and all feedback is welcome.

MPapst, 2022-10-01:
> > what about: Hosting a Modbus TCP or UDP listener inside kestrel using endpoints?
> 
> @MPapst - do you have a use case for this?

The idea when I was posting this was to leverage the power of Kestrel, as it allows hosting of custom tpc/udp based workloads and handles the connections on its own, so there is no need to handle them using a TcpListener. 
I know that there are some examples from David Fowler (@davidfowl).

I recently stumbled over a rust implementation of a Modbus Library which has a very good API, maybe you want to have a look:
https://github.com/stepfunc/rodbus/blob/main/ffi/bindings/dotnet/examples/server/Program.cs
They are using Handlers for Authorization and for Storing of Data.

rquackenbush, 2022-10-04:
@MPapst - I like that rust implementation. I was already thinking of doing something similar (aka a much less dogmatic approach) and this helps.

rquackenbush, 2023-08-21:
Heads up - NModbus 4.0.0-alpha008 is available.  It only supports TCP/UDP transports, but is ready to be abused.