NModbus/NModbus #68: How to communicate with Slave that uses plain .Net Stream as data transfer method?

Hooch180, 2020-07-29:
Hi,

I have device that has a simple USB HID Device interface. In short. My only way to communicate with it is to send/receive bytes via old simple .Net Stream.

When I send manually prepared modbus frame to this device it responds with modbus frame.

I don't want to build frames manually. How would I go about using your library to communicate with this device?

I can't use TPC or SerialPort. Which class/interface do I need to inherit and expand to allow communication with Slave via .Net Stream?

Thank you in advance for your help.

MPapst, 2020-10-31:
Hi,

can't you go with an own imeplementation of `IStreamResource` ? 
https://github.com/NModbus/NModbus/blob/develop/NModbus/IO/IStreamResource.cs

Hooch180, 2021-01-27:
Thanks for suggesiton. It worked.

closed 2021-01-27 by Hooch180