NModbus/NModbus #69: Send response from Slave to Master

redplane, 2020-09-10:
Hi,

I'm working with 2 **Console applications** which are for sending/receiving Modbus RTU Command.
I have read [this document](http://ftp.icpdas.com/pub/cd/8000cd/napdos/modbus/nmodbus/nModbus_API_Manual_v1.2_en.pdf) to know how to implement **ModBus RTU master** and **ModBus RTU slave**.

About **Modbus RTU master** part, everything is fine. The document is very clear.
But I don't see any examples about reponding a message from **Slave** to **Master**.

For example:

Master sends: **01 03 00 00 00 26 C4 10**
Slave will return: **01 03 4C A7 FF 00 38 C2 8F 3C F5 33 33 41 E7 66 66 42 74 3E 14 44 68 00 0C 00 01 66 66 42 56 00 00 42 F7 02 01 00 01 99 9A 41 21 01 43 32 46 41 25 A5 9C 42 CF 00 00 42 70 00 00 42 C8 40 00 45 9C 00 00 43 48 39 9A 45 59 00 44 33 33 42 49 EF E8**

Can I do this with **NModBus** library ? And How ?

MPapst, 2020-10-31:
NModbus uses a Datapoint Store and will respond the data stored in the DataPoint Store on requests from a Client to this.

If you want to have a callback or custom logic called whenever a Request is received, have a look at the SparsePointStorage sample file: https://github.com/NModbus/NModbus/blob/develop/Samples/SparsePointStorage.cs

MPapst, 2020-11-18:
Note, I just read about the manual you've used. That one from 2013. It's about the same library, but in a completely different version. See the [History](https://github.com/NModbus/NModbus/blob/develop/README.md#history) of NModbus in the readme.md of the repository.

redplane, 2020-12-20:
@MPapst , sorry for my late response.
I have tried the example you mentioned, I see the slave respond to the master.
One thing I can see is `StorageOperationOccurred` event was called when the master sent a request to it.
I wonder how I can send a custom response to the master. 

When the master send the command `01030000000305CB` which is about requesting 3 registers from address 0.
With the below code, what I received on master was: `0103060000000000002175`. 
The response was about 2 bit . What if I want to respond as `010303000000<checksum>`. 
What should I do ?

```
           acTechDataStore.HoldingRegisters.StorageOperationOccurred += (sender, args) =>
            {
                //var bytes = BitConverter.GetBytes(10);
                Console.WriteLine($"Holding registers points: {args.Points.Length}");
                Console.WriteLine($"Holding registers: {args.Operation} starting at {args.StartingAddress}");
            };
```

MPapst, 2020-12-21:
That abstraction is done by the library for you.
I haven't digged that deep into it. But the Slave Listener is responding automatically with the response message. The [ModbusRtuTransport is calculating the crc on its own](https://github.com/NModbus/NModbus/blob/04a54c8ddc131e4708bcb0a14871c941b3402243/NModbus/IO/ModbusRtuTransport.cs#L66). 

If you want to change that behavior, you will need to change it deeply in the library.
Is there any reason why you need to do that? In general the library should fulfill low level handling according to the modbus standard.

rquackenbush, 2021-04-20:
Nothing actionable here. Will re-open if you add additional feedback.

closed 2021-04-20 by rquackenbush