NModbus/NModbus #71: ModbusSerialSlaveNetwork.ListenAsync is not async

jim-gruenbacher, 2020-10-21:
`ModbusSerialSlaveNetwork.ListenAsync()` never yields execution with an `await` operator, and will block indefinitely until the underlying serial port is closed.  This can be verified by running the `ModbusSerialAsciiMasterReadRegistersFromModbusSlave()` method in the Program.cs file of the Samples project, which clearly does not work as intended.

The function blocks because it ultimately calls `SerialPort.Read()` which is synchronous and blocks until it receives the number of requested bytes.

I'm currently working around the issue with this very ugly extension method:

`public static async Task ListenWorkAroundAsync(this IModbusSlaveNetwork modbusSlaveNetwork, CancellationToken cancellationToken)
{
    cancellationToken.Register(modbusSlaveNetwork.Dispose);
    Action listen = () =>
    {
        try
        {
            modbusSlaveNetwork.ListenAsync(cancellationToken);
        }
        catch (NullReferenceException)
        {
            System.Diagnostics.Debug.WriteLine("Assuming shutdown of serial port.");
        }
    };

    Task listenCompleteTask = Task.Factory.StartNew(listen, cancellationToken, TaskCreationOptions.LongRunning, TaskScheduler.Default);
    await listenCompleteTask.ConfigureAwait(false);
}`

rquackenbush, 2023-08-22:
I actually just happened to have ordered a number of FDTI RS485 adapters so I could do some dev work on serial. I think they might be sitting on my front porch. I can hopefully get to this in the short term.

niarbeht, 2023-10-13:
If you set a timeout on the serial port, pass a cancellationtoken in, and fire off ListenAsync in it's own thread (there's probably a way to shove it into a task), it can be made to work, but it's pretty clearly not really async.  I'd recommend at a minimum that some documentation gets added to the function to require timeouts on the serial port, or throw an error if it sees that infinite timeout is set on the serial port.

cheahweileow, 2025-01-21:
i have spent full day looking for workaround on this but failed. ListenAsync still blocks despite using cancellation token, thread , runasync or backgroundservice. eventually i revert to NModbus4 and it works