NModbus/NModbus #72: Samples are outdated

kaamil1984, 2020-11-02:
Hi guys,

I have downloaded current NuGet package (3.0.62) and I'm trying to run it on .NET Core 5.0.

I took samples and this does not work at all because there is no such method in factory.

    SerialPort port = ...
    IModbusMaster master = factory.CreateRtuMaster(port);

Could you update samples application please or give some information how to create ModbusRtuMaster on SerialPort in current version?

jgmdavies, 2020-11-02:
I think these days you need to include 'NModbus.Serial' from Nuget, and then do something like:

var adapter = new SerialPortAdapter(serialPort);
result = factory.CreateRtuMaster(adapter);

HTH,
Jim

kaamil1984, 2020-11-02:
Nope. There is no such thing like `SerialPortAdapter` in my 3.0.62 :(

Constructor wants IModbusTransport.

jgmdavies, 2020-11-02:
@kaamil1984 

I'm also using 3.0.62.  Did you install 'NModbus.Serial' (also 3.0.62) from Nuget in your project(s)?

Actually you can now use just:
`result = factory.CreateRtuMaster(serialPort);`

as 'NModbus.Serial' includes an appropriate extension.

kaamil1984, 2020-11-02:
Oh dear. It works. 

**Installing additional package NModbus.Serial from NuGet helped.**

Thank you!

closed 2020-11-02 by kaamil1984

jgmdavies, 2020-11-02:
@kaamil1984

That's good!  Always a shock when things work...