NModbus/NModbus #73: ModbusSlaveNetwork DataStore Data change Events

Andreaxxx, 2020-11-04:
Hi All,

I am currently using the NModbus to simulate a number of Servers (Modbus Slave over TCP). The idea is to have a replicate of real instrumentation, on which I can modify data as I want to in order to provide confirmation that the Modbus Interfaces are correctly mapped on our PLC interface.

The way that I have structured the code is to have the data mapped into the modbusSlaveNetwork object instance (dedicated to the device I want to simulate) and change their values via the associated "Datastore.CoilInputs.WritePoints" to write or "Readpoints" to read back.

However, I am interfacing the data within the Datastore object via a DataGrid Object. My desire is to have the data updated onto the Datagrid at any time values are changing within the Datastore object. Unfortunately though I have not been able to figure out which event (if any) will automatically trigger an event to notify the changes on the Datastore object.

Can you confirm whether I am using the wrong the object (Datastore) or I shall build the code in a different way to allow the events to be triggered accordingly?

thanks!

MPapst, 2020-11-04:
Hi,

the default Data Store does not provide mechanism to update attached Layers. You would need to implement an own DataStore using e.g. an `ObservableCollection<T>` which provides the possibility to subscribe for updates.

As this looks like you are directly binding the DataGrid to the DataStore, in terms of application architecture you should think of a Business Layer between the communication and the gui.

Andreaxxx, 2020-11-05:
Thanks for the answer! It makes sense... I will try to have a look at your proposed solutions.
And also thanks for the Business Layer architecture, I am not familiar with it but at a first glance it seems to be interesting.

closed 2020-11-05 by Andreaxxx

Andreaxxx, 2020-12-10:
Hi there,
sorry for getting back to you again on the same topic... I tried to look into the possibility of using ObservableCollction<T> to event changes. However, I struggled a bit with it.
I cannot find a way to event the Datastore changes, any way you could provide some indication on how to achieve that? any help would be appreciated!
thanks,
Andrea L

MPapst, 2020-12-10:
Hi,

have a look into this implementation: https://github.com/NModbus/NModbus/blob/develop/Samples/SparsePointStorage.cs