NModbus/NModbus #74: Exception Handling

JulianWigman, 2020-11-10:
Hi 
What is the preferred method for handling exceptions with this library.
Obviously try-catch blocks but are there any standard exception codes I can trap from the ModbusFactory.

rquackenbush, 2020-11-11:
Hi @JulianWigman - I'm not sure exactly what you're looking for... Are you talking about runtime Modbus error messages from the network or specifically when creating objects using `ModbusFactory`? 

Any kind of code example would be useful.

JulianWigman, 2020-11-11:
Exactly.  Runtime Modbus error messages.

MPapst, 2020-11-11:
You can find all Exceptions defined (and throwed) by the NModbus library directly in GitHub:
https://github.com/NModbus/NModbus/search?q=Exception

closed 2022-01-27 by rquackenbush