NModbus/NModbus #76: Is Slave ID/Unit ID/Server ID of 0 supported?

WattsC-90, 2020-11-13:
Hi.. just trying to figure out where something is falling over in the code.. Modbus TCP spec defines that an ID of 0 can be used, does the library support this? 
Not sure where I would go in the codebase to find it out.. hunting at the moment.

MPapst, 2020-11-13:
Is there anything that makes you think it may not work?

WattsC-90, 2020-11-16:
no.. i was frantically hunting a bug, since found and fixed, but couldnt find anything that explicitly said it was supported. I had not realised until it was raised as an issue that modbus spec defines 0 as a valid ID so was frantically hunting and trying to cover all bases simultaneously :) thank you for your response. (and sorry for looking like a rank amateur asking if its supported when the bug was in my code!)

closed 2020-11-16 by WattsC-90

MPapst, 2020-11-16:
No worries, Modbus looks easy, but in detail its very strange sometimes :)