NModbus/NModbus #77: .NET project SDKs

JulianWigman, 2020-11-25:
Hi
I tried to switch my VS .NET 4.6 solution to the new .NET CSPROJ SDK format and whilst I still left it targeting "net46" as soon as I am in that format the NMODBUS library complains that it does not support .NET 4.0 and wont compile.  Huh?
Is anyone running NModbus with a new SDK format project?
Regards

rquackenbush, 2020-11-25:
Hmm - haven't tried that. That is the first thing I'm planning on doing for the 3.x version of NModbus.

JulianWigman, 2020-11-27:
Do you have an ETA on testing it or when likely to be available as trying to get a .NET 5.0 application going and project needs to be in SDK format.

MPapst, 2020-11-28:
If I am not missing something, all .NET (Core) based projects are using that new SDK format.
So I am using NModbus in a .NET Core 3.1 context in the SDK style project format without any issue.

rquackenbush, 2022-01-27:
The latest NuGet packages support NET6. Will that work for your current setup?

closed 2022-01-27 by rquackenbush