NModbus/NModbus #79: Add support for Modbus RTU Over TCP

Jah1980, 2020-12-06:
Good afternoon! 
I would like to use this protocol in my work. There are many devices-serial RTU slaves-that are connected to TCP routers. In other words, only the transport layer is converted. 
Is it possible to implement it using your library?
[Modbus RTU Over TCP](https://www.fernhillsoftware.com/help/drivers/modbus/modbus-protocol.html)

jlalik, 2021-01-12:
It can be done using ClientAdapter.

                        _tcpClient = new TcpClient(Ip, Port);
                        _tcpClientAdapter = new TcpClientAdapter(_tcpClient);
                        _master = _factory.CreateRtuMaster(_tcpClientAdapter);

BC-SCI, 2021-05-04:
Is there an example for implementing a Modbus slave over TCP?

rquackenbush, 2021-05-04:
@BC-SCI 

https://github.com/NModbus/NModbus/blob/develop/Samples/Program.cs#L466

```c#
        public static void StartModbusTcpSlave()
        {
            int port = 502;
            IPAddress address = new IPAddress(new byte[] { 127, 0, 0, 1 });

            // create and start the TCP slave
            TcpListener slaveTcpListener = new TcpListener(address, port);
            slaveTcpListener.Start();

            IModbusFactory factory = new ModbusFactory();

            IModbusSlaveNetwork network = factory.CreateSlaveNetwork(slaveTcpListener);

            IModbusSlave slave1 = factory.CreateSlave(1);
            IModbusSlave slave2 = factory.CreateSlave(2);

            network.AddSlave(slave1);
            network.AddSlave(slave2);

            network.ListenAsync().GetAwaiter().GetResult();

            // prevent the main thread from exiting
            Thread.Sleep(Timeout.Infinite);
        }
```