NModbus/NModbus #80: 43 / 14 (0x2B / 0x0E) Read Device Identification

EHoglid, 2021-01-13:
Add function to read Device Identification

https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf    
Page 44
[Modbus_Application_Protocol_V1_1b.pdf](https://github.com/NModbus/NModbus/files/5809409/Modbus_Application_Protocol_V1_1b.pdf)

jlalik, 2021-01-14:
Did You try using ExecuteCustomMessage?

 TResponse ExecuteCustomMessage<TResponse>(IModbusMessage request)
            where TResponse : IModbusMessage, new();

willianSteffler, 2021-08-18:
How could I do this using RtuTransport ?

I wrote a service to handle the custom message code 43. implementing `IModbusFunctionService` The problem is that the frame size is variable as you can see in the picture.
[APPLICATION PROTOCOL SPECIFICATION V1.1b3](https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) 
[image]

Since the response length is variable depending on the field 'Number of objects' and the size of objects in the objects list, i can't determine its total length without reading all the objects list, how could I read the total frame length ?

it seens that we need a 'readAllBytes' in IStreamResource

bdube, 2022-03-02:
Our solution involved `ExecuteCustomMessage` like @jlalik mentions, but that solution involves implementing `IModbusFunctionService` as @willianSteffler mentions. It's the implementation of `IModbusFunctionService.GetRtuResponseBytesToRead` where things get sticky. The method is expected to determine from the `byte[] frameStart` parameter how many more bytes need to be read to finish reading the response. For MEI responses, the answer can't be determined from the four bytes in `frameStart`. The full response can span multiple request-response transactions if it doesn't fit in one frame.

Our solution and implementation of `IModbusFunctionService.GetRtuResponseBytesToRead` was only possible because we also control the MEI implementation on the hardware device. For a general solution, I think NModbus might need to be "MEI aware" and handle this function differently.

pzahra, 2022-12-22:
Yes, that function needs to return not just how many bytes to add to the frame, but also an indication that it's not finished. Most of them will be fine with returning a constant + EOF, but this one will need to be called multiple times as it continues to read more chunks.

Or better yet, pass in a callback to Read(count) so it can simply return the completed frame all concatenated and ready to use in one go.