NModbus/NModbus #82: WritetoHolding register crashes

AnkitaPadma, 2021-01-28:
Hi ,

I m trying to write to holding registers, and if there is a mismatch in the communication settings like baudrate or parity mismatch the app just crashes without throwing any exception. 
How to handle this?
Note have tried try catch, but the call never comes to catch block
try
            {
                master.WriteMultipleRegisters(slaveadd, (ushort)regAdd, data);
            }
            catch(Exception ex)
            {
                MessageBox.Show(ex.Message);
            }

AnkitaPadma, 2021-01-28:
@Fuchks please suggest

Fuchks, 2021-01-28:
Hi @AnkitaPadma,
it could help, if you add the code for establishing connection.
What does "app crashes" mean: unresponsive, vanishes, ...?

AnkitaPadma, 2021-01-28:
@fuchks below is the code to establish the connection.
           port = new SerialPort(comport);

            port.BaudRate = baudrate;
            port.DataBits = databits;
            port.Parity = parity;
            port.StopBits = stopbits;
           
            port.Open();
           
            IModbusSerialMaster master = ModbusSerialMaster.CreateRtu(port);
            
When I say the app crashes, i mean that the application hangs and becomes unresponsive, without any exception being triggered

Fuchks, 2021-01-28:
With connections you should always include a timeout.
For Serial set ReadTimeout & WriteTimeout to appropriate values.

As there is no timeout set, the app waits until it receives a response - which will never occur with wrong settings and seems to hang.

AnkitaPadma, 2021-02-02:
@Fuchks Thanks for the suggestion, I will try it out and update.

closed 2021-04-20 by rquackenbush