NModbus/NModbus #85: RtuMaster ReadWriteMultipleRegisters

onimor, 2021-04-16:
How to make ReadWriteMultipleRegisters work with serial port?
I get: Specified method is not supported

rquackenbush, 2021-04-20:
A few questions:
- What version of .NET are you using (.netcoreapp3.1, net5, etc)?
- Which NModbus serial library are you using?

onimor, 2021-04-21:
net5

NModbus 3.0.62
NModbus.Serial 3.0.62
NModbus.SerialPortStream 3.0.62

System.NotSupportedException
HResult=0x80131515
Message = Specified method is not supported.
Source = NModbus
Stack Trace:
at NModbus.Device.MessageHandlers.ReadWriteMultipleRegistersService.GetRtuResponseBytesToRead(Byte[] frameStart)
at NModbus.IO.ModbusRtuTransport.ReadResponse()
at NModbus.IO.ModbusRtuTransport.ReadResponse[T]()
at NModbus.IO.ModbusTransport.UnicastMessage[T](IModbusMessage message)
at NModbus.Device.ModbusMaster.PerformReadRegisters(ReadWriteMultipleRegistersRequest request)
at NModbus.Device.ModbusMaster.<>c__DisplayClass29_0.<PerformReadRegistersAsync>b__0()
at System.Threading.Tasks.Task`1.InnerInvoke()
at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
at WSAMP.Services.Modbus.ModbusMain.<WriteAsync>d__17.MoveNext() in C:\Users\onimo\source\repos\WSAMP\WsampSln\WSAMP\Services\Modbus\ModbusMain.cs:line 182
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
at WSAMP.ViewModels.WeightWindwosVM.<AllowExit>d__18.MoveNext() in C:\Users\onimo\source\repos\WSAMP\WsampSln\WSAMP\ViewModels\WeightWindwosVM.cs:line 81

This exception was originally created in this call stack:
[External code]
WSAMP.Services.Modbus.ModbusMain.WriteAsync () in ModbusMain. cs
[External code]
WSAMP.ViewModels.WeightWindwosVM. AllowExit () in WeightWindwosVM. cs

rquackenbush, 2022-01-27:
Hmm - is this on Windows or Linux? Either way, if you're using NET5, you shouldn't need `NModbus.SerialPortStream`. That was a stop gap I put in place so that we could run serial ports on Linux before the `SerialPort` support was added back to the runtime.

What does your code look like for constructing the Modbus master?

onimor, 2022-02-01:
@rquackenbush , windows .
```
 _serialPort = new SerialPort
                    {
                        PortName = _modbusParametersDto.ModbusComPortNumber,
                        BaudRate = _modbusParametersDto.ModbusComPortBaudRate,
                        DataBits = 8,
                        Parity = Parity.None,
                        StopBits = StopBits.One,
                        ReadTimeout = 3000,
                        WriteTimeout = 3000
                    };
                    _serialPort.Open();
                    _master = _factory.CreateRtuMaster(_serialPort);
```