NModbus/NModbus #88: Timeout bug with TcpClient

joao147, 2021-07-20:
The bug happens after a timeout in a call of API, for example, I'd like to read a Coil, but a timeout happens, after this timeout error every single call throw exception like this "Response was not of expected transaction ID. Expected 8, received 7", data of first call that was request continue on wait. I tried to clean buffer but it don't work, only reconnecting it was solve, but that's not ideal.

joao147, 2021-07-20:
I inspect on API code and apparently request is create, it's all right, but when response throw error, all are ignored, and when a new request is call it's create a new request that transaction id is different from response, because last response wasn't create, and all new request and response will have different id.

Fuchks, 2021-07-20:
Did you check if transaction id of response fits request at protocol level?
I had a device that responded wrong and only workaround was to establish new connection after an error.

Tbeck-91, 2022-04-08:
I'm also having this issue, if there is a timeout on the request the transaction ids get out of sync

tmoers, 2022-05-24:
I have the same issue. Somehow the ModbusTransport seems to have gotten into an invalid state. My communication gets broken, continuously toggling between the following errors:
- Response was not of expected transaction ID. Expected <n>, received <n-1>
- Maximum amount of data 127 registers. (Parameter 'NumberOfPoints')

mtorre-iot, 2023-12-18:
I am getting exactly the same issue, mostly when trying to write single registers and reading input status. What I am doing is close the client, and try to reconnect, but I am having issues with the Modbus Server, it gets annoyed so many connection and reconnections. @joao147 did you find a workaround for this? If not, what do you suggest the best way to restart that transaction ID count? I look forward to hearing from any of you. Cheers!

LFDCC, 2023-12-28:
I also encountered this problem when reading hold registers, but it didn't happen every time. How to solve it?

Zapnologica, 2024-05-17:
Hey guys anyway to solve this issue?

We are getting lots of errors like:

"Response was not of expected transaction ID. Expected 4, received 1."