NModbus/NModbus #89: write ok,but have error

jackswei, 2021-09-06:
Connect vigor plc rs485,it write ok,but show error
"Unexpected start address in response. Expected 5000, received 0."

rquackenbush, 2021-09-06:
Have you confirmed that the serial port settings are correct? E.g. baud
rate, parity, etc?

Also - what RS-485 transceiver are you using?

jackswei, 2021-09-06:
it's tcp modbus
i use test code 
 using (TcpClient client = new TcpClient("172.16.30.111", 502))
            {
                var factory = new ModbusFactory();
                IModbusMaster master = factory.CreateMaster(client);

                master.WriteSingleRegister(1, 5000, 120);
                var V = master.ReadHoldingRegisters(1, 5000, 1);

            }

rquackenbush, 2022-01-27:
Are you able to verify that the write worked using some other tool or user interface?

closed 2023-08-21 by rquackenbush