NModbus/NModbus #91: factory.CreateRtuTransport((port) ???

wwowa-rus, 2021-10-03:
Не удается привести... Ошибка.
Unable to cast ... Error.
 using (SerialPort port = new SerialPort("COM3"))
            {              
                var factory = new ModbusFactory();
              IModbusRtuTransport transport = factory.CreateRtuTransport((port);
.....
           }
**Аргумент 1: не удается преобразовать из(Argument 1: Cannot convert from) "System.IO.Ports.SerialPort" в "NModbus.IO.IStreamResource".**	
И где пространство имен NModBus.Serial  ???	
And where is the namespace NModBus.Serial ???

MPapst, 2021-10-03:
NModbus.Serial is a seperate package https://www.nuget.org/packages/NModbus.Serial/

On which line are you getting which Exception?

wwowa-rus, 2021-10-04:
I installed NModbus.Serial and everything works. Thanks.

MPapst, 2021-10-04:
Can you please close the issue?

closed 2021-10-05 by wwowa-rus