NModbus/NModbus #94: Size of lazy Points array should be ushort.MaxValue + 1

ShashikkYoko, 2021-12-16:
Otherwise we are short of one register for both Holding Registers and Input Registers.

rquackenbush, 2021-12-16:
Quite right! Good catch.

[image]

0xFFFF = 65,535

`const UInt16.MaxValue = 65535;`

Here is some code that demonstrates the problem:

https://dotnetfiddle.net/3BYyp1

ShashikkYoko, 2021-12-16:
Thanks for a quick response.
When will the change be reflected in the latest Nuget package? 
I saw the code change already, but wondering when will the new formal tested build be available in Nuget?

Regards
Shashi

rquackenbush, 2021-12-16:
I'm working on getting this built. There are some problems on my AppVeyor
configuration that I'm working through.

ShashikkYoko, 2021-12-20:
Hope you had a chance to fix the AppVeyor configuration. Your help is much appreciated.

rquackenbush, 2021-12-20:
Almost there. It looks like all that remains is sorting out a new api key
for nuget. I’m
Hoping to have that done today.

rquackenbush, 2021-12-20:
Done. Version 3.0.66 has been deployed. Please let me know if you have any further issues or questions.

closed 2021-12-20 by rquackenbush