NModbus/NModbus #95: Read Holding Regist Problems

jot4p, 2021-12-17:
Hi, 

The value readed from address 0 is not correct for one device.

For the same device and same settings:

[image]

The correct value from address 0 is -21620 ... what is the problem ? 
[image]

NModbus 

[image]

rquackenbush, 2021-12-22:
This is an interesting case. Is this TCP or serial based? If it’s TCP based are you able to get a wire shark capture of each client and attach it here forms to look at?

jot4p, 2021-12-22:
> This is an interesting case. Is this TCP or serial based? If it’s TCP based are you able to get a wire shark capture of each client and attach it here forms to look at?

Sorry, my fault.. confusion between input regists and holding regists... 

problem solved. 

Thanks

closed 2021-12-22 by rquackenbush