NModbus/NModbus #98: Invisibility of NModbus.Data.RegisterCollection class

DimaCalugari, 2022-01-20:
If you want to create a custom modbus message and you must initialize some specific data, you cannot access the `NModbus.Data.RegisterCollection` class because it is set as `internal`. A `public` access modifier instead would help a lot.

code4freee, 2022-01-20:
That's right... 
In NModbus4 this class was public, now internal.
For some reasons we have to send function code 100 which is not specified in modbus spec.
If access to internal class, as proposed above, would be granted it would help us very much to change from NModbus4 to actual implementation.

DerDan, 2022-01-27:
i also need this

rquackenbush, 2022-01-27:
I've made this class public and it is available as of version 3.0.68 of the NuGet package.

closed 2022-01-27 by rquackenbush