NModbus/NModbus #99: Network Monitoring

cdrose, 2022-01-24:
I am looking to build a simple monitoring utility for debugging a modbus RTU network. The code will need to act as a silent slave which accepts and decodes any modbus communication on the bus for display but never responds. Is there some way to implement this in NModbus? I am guessing it would be a custom implementation of an RTUSlaveNetwork, but with most classes being internal I cant immediately see a way to do this. Is there any way to build something like this using the library?

rquackenbush, 2022-01-27:
I've made several attempts at doing exactly this, and none of them have gone particular well. Here is a repo that includes some work in that area:

https://github.com/rquackenbush/ModbusRegisterViewer

The challenge I ran into was that windows isn't a real time environment, and Modbus depends on timing for encoding / decoding. There are tricks you can attempt (e.g. examining a stream of bytes for a message where the calculated CRC matches, but this happens in random streams of data more often than you would expect and leads to many false positives). 

I'm sure that there is a way to do it, but I haven't been successful quite yet. If you find more success, please share your results!

cdrose, 2022-01-28:
I see your point. Using a PC for a master is relatively straight forward because the master controls when messages are sent and received. A PC as a slave isn't too bad if you know to only look for messages addressed to you. But PC as a silent slave that has to catch all messages is a much more difficult problem.

rquackenbush, 2022-01-28:
I think you're on the right track. It's a fuzzy / fiddly endeavor, but I think is possible with a smart algorithm like you're suggesting.

0xStuart, 2023-07-30:
Just ran into the same thing.  
What I have done is fill up a byte buffer of modbus data, and then using a pair of pointers in nested for-loops, look for matching CRCs.

Before you start burning up your cpu with each scan, check the second byte against this:
```
static bool IsStandardFunctionCode(byte code)
    {
        switch (code)
        {
           case 01: // Read Coils
           case 02: // Read Discrete Inputs
           case 03: // Read Holding Registers
           case 04: // Read Input Registers
           case 05: // Write Single Coil
           case 06: // Write Single Register
           case 07: // Read Exception Status
           case 08: // Diagnostics
           case 11: // Get Comm Event Counter
           case 12: // Get Comm Event Log
           case 15: // Write Multiple Coils
           case 16: // Write Multiple Registers
           case 17: // Report Slave ID
           case 20: // Read File Record
           case 21: // Write File Record
           case 22: // Mask Write Register
           case 23: // Read/Write Multiple Registers
           case 24: // Read FIFO Queue
               return true;
           default:
               return false;
        }
    }
```
It will make sense if you read the RTU spec. Once your locked on, if you wanted to get fancy, you can start decoding the data, and deduce where the CRC should be.  No more need for timing (whose idea was that?  madness!)