NModbus/NModbus #108: Unclear 'startAddress' argument name/documentation for IModbusMaster methods

mycroes, 2022-04-05:
Almost all `IModbusMaster` methods use a `startAddress` parameter to indicate the starting offset for the operation. As [this document](https://www.controltechnology.com/Files/common-documents/application_notes/IMPORTANT-NOTE-CONCERNING-ADDRESSING-IN-MODBUS-NET#:~:text=If%20the%20Master%20does%20not,be%20converted%20to%20address%20101.) points out, there's a difference between what is commonly referred to as the address of a register and the offset that's sent over the wire.

To remedy this issue I propose to at least document all `IModbusMaster` methods `startAddress` (or derived) arguments with the fact that it's the zero-based offset and that zero in this case represents memory address 1. I'd be happy to do a PR for this change if that's a desired solution.

To further highlight this difference I think it would also be a good idea to change *address* to *offset* in all comparable argument names (I guess that excludes `slaveAddress`, because it's usage isn't comparable). I think this change might at least provide a trigger for users that there might be a difference between address and offset, which probably is the most common situation.

MPapst, 2022-04-06:
I've worked with a couple different modbus implementations through device and scada vendors.
The Point you are describing is interpreted different across those vendors, which is a painpoint overall.

Nevertheless, changing the behavior would lead to a breaking change and will make it impossible to address the first register for a device that is using zero based approach.

mycroes, 2022-04-07:
I only suggest to change the documentation and perhaps the name, not the actual implementation. I agree that would be a bad idea.